Reedyuk/blue-falcon #170: No way to clear dead peripherals

mcgirr5799, 2025-05-09:
I've noticed that when I collect the peripheral flow it seems to keep holding dead devices. Currently I see no way of clearing these ghost devices.

Is there a solution to this or is it a unimplemented feature. It's currently messing up my looking for device list as some are still off.

Reedyuk, 2025-05-10:
Yes its quite likely i haven't implemented a clean up for dead peripherals.

What would you expect to class as dead peripherals?

mcgirr5799, 2025-05-10:
One that drops devices that haven't been seen in N milliseconds.

Reedyuk, 2025-05-10:
Tricky one, as this is subjective and different on all platforms.
I guess we would need to have a preset time that users could override.
The other issue is platform, not sure we get an OS level callback for when a device is not broadcasting anymore(for android, ios, js and java)

mcgirr5799, 2025-05-12:
Would it be appropriate to have a clear peripheral list command, meaning we can just call it every N seconds, clearing the list and it repopulates with new found devices