Reedyuk/blue-falcon #172: BlueFalconDelegate doesn't work on iOS if BlueFalcon is initialized in a library instead of the main application

tylerdandandan, 2025-06-19:
I'm passing the Android Context and UIApplication.sharedApplication via Koin. Previously, my Bluetooth logic was in shared code and everything worked correctly. Now I've moved the Bluetooth logic to a separate KMP library — everything works fine on Android, but on iOS, BlueFalconDelegate is not being called (permissions are granted, and the scanner correctly discovers devices)

Device: iPhone X
BlueFalcon version: 2.3.0

Reedyuk, 2025-06-20:
Thats odd because thats how i have used blue falcon in my projects and it has worked fine.
Could i ask you to share an example project of this?

tylerdandandan, 2025-06-20:
@Reedyuk In my case the BlueFalcon API was created twice. When I made sure there was only one instance everything started working. BlueFalcon is supposed to be a singleton?

Reedyuk, 2025-06-20:
Also if you have gone to the effort of using koin, it should be relatively trivial to make it a singleton. You might need a layer of abstraction but otherwise its easy

Reedyuk, 2025-08-10:
Closing due to its intended purpose to be a singleton.

closed 2025-08-10 by Reedyuk